pstolarz/OneWireNg #2: reset after power on

adbensi, 2020-02-19:
Hi!

 I have one hardware that is working, and I build the Dallas example to know how this new lib work.

 But, on the load #include "OneWireNg_CurrentPlatform.h" I get reset.. on the ESP8266.
 Cause 1: normal boot after power-on

 I define #define OW_PIN            13 and build it.. only this.
 I put many DEBUG on the first line of the setup, but it don`t show, then, I think the problem is on the  "OneWireNg_CurrentPlatform.h".

 Please, What I can do?

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vb3acf27f
~ld
⸮bl`r⸮nl
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vb3acf27f
~ld
⸮bl`r⸮nl
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vb3acf27f
~ld
⸮bl`r⸮nl
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vb3acf27f
~ld

pstolarz, 2020-02-20:
@adbensi Thanks for the feedback. You probably found a severe bug. Will be fixed soon.

adbensi, 2020-02-20:
Hi!
If I can do something, please tell me.. I made another sample with no block
loop and detect sensor ID too, If you like to extend samples code

Thanks a lot!!

pstolarz, 2020-02-20:
@adbensi @MHz000 Could you check my last commit on master with fix for the ESP8266 crash. I'd be grateful for test with and without internal pull-up resistor configured - last argument of the `OneWireNg_CurrentPlatform()` constructor set to `true` / `false`. Let me know about results...

MHz000, 2020-02-21:
Bingo!
Chapeau, chapeau!
After changing to
static OneWireNg *ow = new OneWireNg_CurrentPlatform(OW_PIN, PWR_CTRL_PIN,
*true*);
#else
static OneWireNg *ow = new OneWireNg_CurrentPlatform(OW_PIN, *true*);
no crash any more (while set to false system crashed)
arduino serial monitor reads:
----------
28:49:49:CC:3:0:0:28 -> DS18B20
  Scratchpad: 73:1:4B:46:7F:FF:D:10:41
  Temp: 23.187 C
Congratulations!

What next? Checking the DS2438? May I ask you for an strip down example to
run the check I'll be glad to support developent of your lib. Stripped down
version means: no parsite power, no address checking, just one device on
the wire. Keep it simple as it can be makes error checking easy.
Jochen

pstolarz, 2020-02-21:
Thanks @MHz000 but it looks like you're still working on 0.4.0 version. Could you confirm that on the latest master commit both variants work fine?

MHz000, 2020-02-21:
downloaded from github, include .zip lib from downloads, compiling the
example
=> Error compiling for board Generic ESP8266 Module.

:-)

my fault??

adbensi, 2020-02-21:
> @adbensi @MHz000 Could you check my last commit on master with fix for the ESP8266 crash.

@pstolarz No more reset in the false condition :)

But I no longer have the hardware to run the long test.

I will assemble a new unit exclusively for this purpose.

Some sensors, independent of the library used, work more properly when the power supply is interrupted before starting the reading continuously. I am considering this in my layout .. the point is that I have not found another way to make a sensor work if it doesn’t start on the first attemp. Then this harware fix solve theses issues.

 May be bad sensor?

 I have a lot of sensors that don´t work .. without stamp brand/model, did You hear about this?

 I'll be glad to support developent of your lib too.

 I put one file to test it..
 https://github.com/adbensi/OneWireNg/pull/1/files#diff-acd88268549e9aa747404ae7c530abcd
Best Regards!

pstolarz, 2020-02-21:
> Some sensors, independent of the library used, work more properly when the power supply is interrupted before starting the reading continuously. I am considering this in my layout .. the point is that I have not found another way to make a sensor work if it doesn’t start on the first attemp.

Could you check my updated Dallas_Temperature example (last master commit) if it fixes the issue?

adbensi, 2020-02-21:
Yes, I will do it, I want to understand why too.
I,ll be back :D
Thanks a lot!

pstolarz, 2020-02-22:
I'm closing this issue, since the problem it concerns looks to be fixed. I created a new version of the library 0.4.1 with the bug fixed. Thanks guys for the report and tests.

Piotrek

closed 2020-02-22 by pstolarz