pstolarz/OneWireNg #3: crc16 check

lukjj, 2020-02-22:
Hi,
can you please add crc16 function. I'm trying create onewire arduino nano to nano communication by  [OneWireHub](https://github.com/orgua/OneWireHub) and this libary. On slave side I was select bae0910 protocol and this need crc16. With [Onewire ](https://github.com/PaulStoffregen/OneWire) libary I have some problem with more data communication on nano cpu, but your onewireng work perfect in first tests. 
In [Onewire ](https://github.com/PaulStoffregen/OneWire) libary is crc16 working. 
Thanks
Regards

pstolarz, 2020-02-23:
Ok, I will add. What kind of problems do you experience with OneWire?

lukjj, 2020-02-24:
Thank you.
Probably cpu speed. On onewire hub is "first bit of byte timeout" error when I used onewire libary. Then there was not answer to master (0xff unchanged).
I think mega328p have too long waiting somewhere. It is on both sides and it can bring some big delay together. I had only communication code there not more.
With DS18x20 emulator was not problem I think (I was test more combinations and Im not sure now ), but another examples have this problem (DS2408,DS250x). Problem can be with sending more bytes.
I have no osciloscope to measure something and also I was not look too deep to code to find exact problem. 
Same slave and your libary works (memory reading and writing commands).

pstolarz, 2020-02-24:
Probably, but I don't think an emulator is a good mean to assess a library. It is not a real device and may provide its own bugs/timings too... In fact I'm using slightly different timings than OneWire, but it shouldn't be a big deal I suppose. In my lib write-1 is virtually the same as read (aka touch-1), with exactly the same timings. In OneWire they are different with slightly different timings.
And just for curiosity, how many errors (in percents) you get with OneWire and how many with mine?

pstolarz, 2020-02-25:
CRC-16/ARC added. Need to be enabled via `CONFIG_CRC16_ENABLED` in the library config file `OneWireNg_config.h` (disabled by default). I additionally added generic CRC-8/16/32 computation routine (implemented by C++ function template). May be useful to calculate arbitrary CRCs with different polys.

Enjoy

closed 2020-02-25 by pstolarz

lukjj, 2020-02-25:
For onewire I have always errors and yours without (now only short cable - later will be around 40m old 230V cable). 
CRC 16 looks working. Perfect work, thank you a lot.