pstolarz/OneWireNg #4: Add overdrive mode support

cactorium, 2020-04-15:
Hello! Would you guys be up for adding support for an overdrive mode? I'm working with a chip that uses only this high speed version of 1-Wire, so I could work on the initial implementation

pstolarz, 2020-04-16:
I was considering OD mode, but since I don't have a device for tests I did't decide to implement it. A possible issue I see with the mode it has [strict and short time intervals](https://www.maximintegrated.com/en/design/technical-documents/app-notes/1/126.html) for 1-wire bus activities (e.g. 2.5 usec). It may be problematic to implement in an elegant and platform agnostic way in the library. But if you have OD-ready device and are eager to help we may give it a try.

pstolarz, 2020-04-18:
1st drop of the OD support was added to [`overdrive`](https://github.com/pstolarz/OneWireNg/tree/overdrive) branch. Check [Overdrive_Mode](https://github.com/pstolarz/OneWireNg/blob/overdrive/examples/Overdrive_Mode/Overdrive_Mode.ino) example and let me know about results.

cactorium, 2020-04-21:
Wow that was fast! I'll try to test it out some time this week, there's a project taking priority for now

pstolarz, 2020-05-26:
OD mode support added by the latest master commit. Checked on AVR and ESP32 platforms.

closed 2020-05-26 by pstolarz