pstolarz/OneWireNg #5: Please, add 1-Wire emulation capability

woodlist, 2020-05-07:
Recently did a try to transfer analog signal's readout through harsh electromagnetic environment, via UART, by one phisical wire, from master microcontroller Tx to slave micro's Rx. The communication works normal, but time to time the transfer falls. I would be perfect to have a chanse to use 1-Wire protocol for some measurement guarantied delivery to MC.

pstolarz, 2020-05-07:
Sorry, I don't understand what is a correlation between topic of this issue and it's description. Also note, the 1-wire protocol provides CRC as a mean to detect error during transmission. In case of transferred data inconsistency the transfer should be repeated.

woodlist, 2020-05-07:
You are absoulutey right. The subject is not issue, but improvement. As a proffi, who has deep knowledge in 1-Wire technology, this improvement implementation (I imagine so) may require a liitle efforts, to get done by your side. Am I wrong, dear?

woodlist, 2020-05-07:
The subject, am talking about, is made on hardware and commecrially available here: https://www.mikroe.com/uart-1-wire-click
But I suspect, this can be done on software level too.

pstolarz, 2020-05-07:
The HW you mentioned is an UART/1-wire translator basing on DS2480B IC. Writing an emulator of the translator would be possible but it's of of the scope of this library. The library is intended to provide basic 1-wire features which may be leverage to develop other 1-wire related software including the emulator you pointed out. If you have programming experience I may assist you to write the emulator basing on this lib.

pstolarz, 2020-05-07:
You may use this [doc](https://datasheets.maximintegrated.com/en/ds/DS2480B.pdf) for the purpose of the implementation.

woodlist, 2020-05-10:
Very well solution was found at: https://electronics.stackexchange.com/questions/169315/low-cost-simple-communication-between-two-microcontrollers/169325#169325?newreg=db2d136748564670824758ba1e591d8a

closed 2020-09-15 by pstolarz