pstolarz/OneWireNg #6: ExternalPullup

ktannady, 2020-09-15:
does this [DallasTemperature fork](https://github.com/pstolarz/Arduino-Temperature-Control-Library/tree/OneWireNg) 's ExternalPullup example does the exact thing as the parasite powering section [here](https://github.com/pstolarz/OneWireNg#parasite-powering)?

do i need to use the DallasTemperature ExternalPullup example, or the one at the parasite powering section?
or even both of them?
I need to control the ExternalPullup mosfet but i dont know which way to use it properly..

regards,

pstolarz, 2020-09-15:
There are 2 alternative ways to do this:
1. Use the  DallasTemperature fork as you mentioned. Set up `ONE_WIRE_BUS` and `ONE_WIRE_PULLUP` (lines 4,5) pins for your platform in the [ExternalPullup example](https://github.com/pstolarz/Arduino-Temperature-Control-Library/blob/OneWireNg/examples/ExternalPullup/ExternalPullup.ino).
2. Use my [DallasTemperature example](https://github.com/pstolarz/OneWireNg/blob/master/examples/DallasTemperature/DallasTemperature.ino). Define `PARASITE_POWER` (line 21) and set up `OW_PIN` and `PWR_CTRL_PIN` (lines 18, 26) pins for your platform in the source code.

I haven't tested 1st alternative therefore can't guarantee it works. But the 2nd approach was testes and worked on my setup, so I recommend it for the 1st try.

ktannady, 2020-09-16:
Hi, thanks for your reply.
may i know what kind of P-Channel MOSFET you are using on your setup?

regards,

pstolarz, 2020-09-16:
Type of mosfet shouldn't be an issue here. Why did you connect pullup between gate and Vcc? It doesn't make sense. Schematic should be exact as in the readme. BTW 100k pullup is too large. 1-5k is sufficient in most cases.

closed 2020-10-17 by pstolarz