pstolarz/OneWireNg #7: OneWireNg_Wrapper provided by OneWire.h in OneWireNg?

mairas, 2021-04-05:
I began to wonder: instead of having `OneWireNg_Wrapper` in a `DallasTemperature` fork, should it rather be provided by `OneWire.h` in the `OneWireNg` repo? If I understand correctly, in that case the changes to `DallasTemperature` would be limited just to the dependency declarations in `library.json` and `library.properties`?

It might even be possible to eventually convince the PlatformIO crowd to implement a "provides" or "replaces" field in `library.json`, similar to the Debian packaging `Provides` and `Replaces` keywords. In that case, no forking of any downstream repos would be required.

pstolarz, 2021-04-05:
That's good idea and I'd been considering it while porting DallasTemperature. But decided to use the wrapper mainly for these 2 reasons:
* It was simpler and faster to implement. The wrapper implements only subset of the original `OneWire` class required for DallasTemperature library.
* I'm not considering `OneWire` as a mature 1-wire service library with API interface worth to follow. The interface itself has several issues like `power` arg. in `write()` or strange `begin()` method. Moreover implementing the "write-power" functionality via `OneWireNg` exactly as in `OneWire` is impossible due to its buggy character. I'm not going to go into details but parasite powering in `OneWire` as well as high-state bit-banging is one big bug in `OneWire` and it was the main reason I decided to write this lib. If you would like to know more details look at my [original PR](https://github.com/PaulStoffregen/OneWire/pull/8) for `OneWire` dated on Dec 2015.

On the other hand maybe that's not bad idea to write some sort of interface layer between these two libs. I'm aware of popularity of `OneWire` and its broad usage, so providing such `OneWire` to `OneWireNg` interface would help other people experiencing issues with `OneWire` to switch into this lib w/o any additional effort. In spare time I'll consider writing such class and when ready will ask you for test on your platform(s).

pstolarz, 2021-04-09:
@mairas  I've added OneWire compatibility interface into the library as described above. New implementation already on `master` branch (no new lib version yet). [DallasTemperature](https://github.com/pstolarz/Arduino-Temperature-Control-Library/tree/onewire) fork is also updated but on branch `onewire`.

I've checked the changes on Atmega328P (Arduino Uno) and works fine. Could you check the change on your ESP setups?

mairas, 2021-04-10:
I'm happy to report the new compatibility interface seems to work fine.

I changed DallasTemperature to point to `https://github.com/pstolarz/Arduino-Temperature-Control-Library.git#onewire` and OneWireNg to point to `https://github.com/pstolarz/OneWireNg.git`. I then cleaned the PlatformIO `.pio` build tree and let it rebuild the project, checking that DallasTemperature and OneWireNg were actually the new versions. Build seemed to go through fine. I then added the redundant `#include <onewire.h>` entry back to the code and it still worked fine.

Thanks for the effort!

pstolarz, 2021-04-10:
Good to hear. I raised lib version to 0.7.2. Once indexed by Arduino / PlatformIO lib-repos I will update DallasTemperature fork on branch OneWireNg to point to this new version and will ask you to make the test one more time.

Thanks for the good idea of improvement and the test!
Kiitos!

pstolarz, 2021-04-12:
Done. DallasTemperature fork on `OneWireNg` branch is now a clone of the upstream with library descriptors as the only difference.

closed 2021-04-16 by pstolarz