pstolarz/OneWireNg #17: ESP32 RISCV-C platforms support breaks espressif32 builds

mairas, 2021-09-02:
Hi,

the "ESP32 RISCV-C platforms support" commit (e862d5f338afabf21572de7847e56ea81531e00a) unfortunately breaks the espressif32 platform builds:

```
In file included from .pio/libdeps/esp32dev/OneWireNg/src/OneWireNg_CurrentPlatform.h:36:0,
                 from .pio/libdeps/esp32dev/OneWireNg/src/OneWire.h:19,
                 from .pio/libdeps/esp32dev/DallasTemperature/DallasTemperature.h:25,
                 from src/sensors/onewire_temperature.h:6,
                 from src/sensors/onewire_temperature.cpp:1:
.pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoESP32.h: In member function 'void OneWireNg_ArduinoESP32::initDtaGpio(unsigned int, bool)':
.pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoESP32.h:21:33: error: request for member 'val' in 'GPIO.<anonymous struct>::in', which is of non-class type 'volatile uint32_t {aka volatile unsigned int}'
 # define REG_GPIO_IN_LO GPIO.in.val
                                 ^
```

pstolarz, 2021-09-02:
Interesting, all 3 types of ESP32 compiles successfully on my Arduino IDE 1.8.13. ESP32-C3 platform has been tested by other library user (see #13) and worked fine.

pstolarz, 2021-09-02:
#18 fixes ESP32 build for ESP32-Arduino 1.x SDK. I tested if on my ESP32-DevKitC and worked fine (for the test I used PlatformIO IDE). I also tested (on the same setup) for ESP32-Arduino 2.x with Arduino IDE 1.8.13 and looking good. S2 and C3 also compiles fine for 2.x SDK.

Could you test on your setups?

mairas, 2021-09-02:
Yup, tested and commented on the PR. Works a-OK! Thanks!

PR #18 (2021-09-02): ESP32: Arduino SDK compatibility fix
Fixes #17
closed 2021-09-02 by pstolarz