pstolarz/OneWireNg #19: DS18B20 on multiple pins?

VladimirVecera, 2021-09-07:
What would the onewire code look like if I have a DS18B20 thermometer connected to multiple pins?

pstolarz, 2021-09-07:
If you use OneWire lib and it worked fine until ESP Arduino 2.0 simply replace OneWire to OneWireNg. Use [this fork of DallasTemperature](https://github.com/pstolarz/Arduino-Temperature-Control-Library) instead of the original one. This should fix issues.

VladimirVecera, 2021-09-07:
> Pokud používáte OneWire lib a fungovalo to dobře, dokud ESP Arduino 2.0 jednoduše nahradí OneWire OneWireNg. Použijte [tuto vidlici DallasTemperature](https://github.com/pstolarz/Arduino-Temperature-Control-Library) místo původní. To by mělo vyřešit vaše problémy.

If I understand that, I have an existing OneWire, which newly displays an error on ESP32 (2.0.0), I have to delete it and install OneWireNG.
But I already use your library, so is it necessary to change the OneWire library to OneWireNG? I'm so sorry, but I'd rather write. https://github.com/pstolarz/Arduino-Temperature-Control-Library

pstolarz, 2021-09-07:
By replacing the libs by my versions you will be use no conflicts appears.

closed 2021-09-07 by pstolarz