pstolarz/OneWireNg #20: ds18b20 error compilation

DanSkareda, 2021-09-13:
When I compiled example for DS18B20 from OneWireNg in this spring (for Attiny3216), everything was fine. Now it ends up compiling with an error. Can you figure out why? Some bug in the library?

C:\Users\radek\AppData\Local\Temp\cccalydd.ltrans0.ltrans.o: In function `UartClass::begin(unsigned long, unsigned int)':
<artificial>:(.text+0x8ee): undefined reference to `operator delete(void*, unsigned int)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Nastala chyba při kompilaci u desky ATtiny3216/1616/1606/816/806/416/406 w/Optiboot.

pstolarz, 2021-09-13:
`UartClass::begin()` is your code? How it looks like?

uzi18, 2021-09-13:
Please post also core name and version eventually link to core and sketch with error.

DanSkareda, 2021-09-13:
I did a clean install of the Arduino IDE 1.8.16. Using the "json" link I installed the current megaTinyCore from SpenceKonde.In the sketch folder <libraries> I copied the unzipped current version of OneWireNg.  In its Examples folder is the DallasTemperature sample project. I opened that and did a compilation for attiny3216/Optiboot (it doesn't work in general for any circuit). The compilation ended with errors. In the spring I also tried it for attiny3216, everything was fine then. I don't know where the problem is. Maybe in the core, maybe in the library...

pstolarz, 2021-09-13:
Some insight into this issue is [here](https://forum.arduino.cc/t/undefined-reference-to-operator-delete-void-unsigned-int/620428). What is the json link you use to install the core?

pstolarz, 2021-09-13:
This is an issue in megaTinyCore core. I compiler it by adding this lines at the end of `megaTinyCore/hardware/megaavr/2.4.2/cores/megatinycore/new.cpp`:
```
void operator delete(void *ptr, unsigned int) {
  free(ptr);
}

void operator delete[](void *ptr, unsigned int) {
  free(ptr);
}
```

also add these declarations to `new.h` (in the same dir):
```
void operator delete(void *ptr, unsigned int);
void operator delete[](void *ptr, unsigned int);
```

After this the example compiles successfully. Please report this as a bug to [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) and provide link to this issue here.

DanSkareda, 2021-09-15:
That's great. It's working. I followed your advice and modified the core. Then I compiled the project without errors and Dallas DS18b20 is working. :-) Thanks a lot.

> 
> This is an issue in megaTinyCore core. I compiled it successfully by adding these lines at the end of `megaTinyCore/hardware/megaavr/2.4.2/cores/megatinycore/new.cpp`...............

pstolarz, 2021-09-15:
You are welcome. Finally @SpenceKonde will need to look at this and fix it in the core.

closed 2021-09-15 by pstolarz

SpenceKonde, 2021-09-15:
Hm, we already have 
```
void operator delete(void * ptr);
void operator delete[](void * ptr);
```

Are those correct and we also want the versions that also take an unsigned int? What does the unsigned int do?  It looks like a dummy argument than will never be used.... doesn't that generate compiler warnings if we don't give the second argument the `unused` attribute, or am I missing something here? That the core compiles without warnings is a requirement for all releases. 
This is the second time that issues with "missing" lines have been reported, so I'd like to make sure that file has everything it ought to and nothing it oughtn't.. Where is the reference for what should be defined here., and  the documentation for it which explains what these operators do? I will freely admit that I am not particularly competent regarding most of the stuff that's in  cpp but not c. I mostly write pure C at a very low level, and am comfortable with inline assembly, but as soon as classes or malloc get involved, things seem to get a lot more confusing....

pstolarz, 2021-09-15:
I didnt go into details, but it looks like a new gcc changed something in its includes and it ends up by compilation error in the core. This is the information I got from the link mentioned above. It also contains a reference to C++ spec. regarding this extra arg.

If you would need some more detailed insight into this issue I may look at this topic more thoroughly but in the next week.

SpenceKonde, 2021-09-16:
Thank you, the cppreference article there has definitely granted some insight on why this became a problem when it previously wasn't.

pstolarz, 2021-09-24:
Root cause of the issue is described [here](https://github.com/SpenceKonde/megaTinyCore/issues/528#issuecomment-926565116). Library fix provided by #21.