pstolarz/OneWireNg #34: 0.10.2 failing

homonto, 2022-03-08:
Hi,

So after installing 0.10.2 over 0.10.1 I started having totally inconsistent reading.
I have 3 DS18B20 and so far all was OK.
The moment I upgrade the program has issues with finding sensors, or, invalid reading 
Actually I am not able to say which one is when.
As per your example, I get either:
`Invalid CRC!`
or simply the sensor is omitted

Reverting to 0.10.1 solves the problem.
I did not change anything in my code, so maybe 0.10.2 has some breaking changes?

pstolarz, 2022-03-08:
What is your platform?
Change:
```c
# define CONFIG_BITBANG_TIMING TIMING_RELAXED
```
to
```c
# define CONFIG_BITBANG_TIMING TIMING_STRICT
```
in file `src/OneWireNg_Config.h` and let me know about results.

homonto, 2022-03-08:
it works with this setting
with relaxed it is... too relaxed ;-)
My system: ESP32-S2, IDE: Arduino, 3 sensors DS18B20 with 3 wires and 1 resistor connected to GPIO10 on ESP.

Let me know if you need more info.
Since these 3 sensors were measuring the temperature of my Power Supply, since today morning my Power Supply is not working (over/under temperature protection is based on them) - I need to find a way how to handle false alarms like this ;-)

pstolarz, 2022-03-08:
I tested the relaxed timings on my ESP32 devkit (legacy version) and worked fine. I have some S2 board and will check the relaxed timing soon. I'm suspecting you use WiFi communication extensively on your setup with 1-wire sensors read-outs and this causes the problem you observed (WiFi interrupts re-entrant time critical 1-wire bit-bangs).

I will change the timing to strict, thanks for the catch.

> Since these 3 sensors were measuring the temperature of my Power Supply, since today morning my Power Supply is not working (over/under temperature protection is based on them) - I need to find a way how to handle false alarms like this ;-)

Do you mean I spoiled something? ;)

homonto, 2022-03-08:
;-) don't worry - this Linear Power Supply is 5V 10A for my Raspberry Pi farm. I upgraded libs today and it automatically went to 0V due to over temperature. It took me some time to diagnose, especially that Arduino is caching the compiled libs so simple downgrade of library was not successful - I had to ... you know - this and that. But finally I narrowed the issue to be the upgrade so all is back normal - _"u mnie TERAZ dziala"_ as admin would say ;-)

homonto, 2022-03-08:
With regards to WiFi - the power supply provides data to Home Assistant over MQTT so yes, WiFi always ON and MQTT on top of that. I could try to turn off WiFi fo the time when library is measuring - I will try. But it will then take time to reestablish connectivity - I will test and report sir!

pstolarz, 2022-03-08:
Dzięki i fajnie, że rodacy też używają mojej biblioteki. Pozdrawiam ;)

homonto, 2022-03-08:
> Dzięki i fajnie, że rodacy też używają mojej biblioteki. Pozdrawiam ;)

I usually use SHT31 for my sensors, but since I needed more of them I decided to use DS18B20. I started with the standard onewire - NOT WORKING AT ALL! Yours is superb - before upgrade I mean ;-)

homonto, 2022-03-08:
Now working again!
Thx.

[image]
[image]

closed 2022-03-08 by homonto

pstolarz, 2022-03-08:
This is with latest master patch, right?
BTW: What is this soft for controlling the power-supply you use?

homonto, 2022-03-08:
I did not apply the patch yet - I understood the only change is what you already suggested, right?
The entire software is my "wypociny" with ESP32.
The screenshot is from Home Assistant.

pstolarz, 2022-03-08:
Nie bądź taki skromny :-)

homonto, 2022-03-08:
jak mi dasz maila to pogadamy przeciez bo widze, ze rodak z Krakowa ;)

pstolarz, 2022-03-08:
pstolarz@o2.pl

uzi18, 2022-03-08:
@pstolarz  Pewnie że używają :) 

Waiting for platform.io update for this lib. 
Got strange behavior on esp32 - method begin() from dallastemperature scans bus and found device. 
But... Fetched temp reports device as disconnected.
Need more investigation here.

uzi18, 2022-03-08:
@homonto lepiej kanal na gitter.im zalozyc

pstolarz, 2022-03-08:
@uzi18 Check the fix with strict timings and let me know about result. Looks like moving to relaxed timing in 0.10.2 was not clever move (I checked it on my ESP32 and worked fine but didn't launch extensive tests with WiFi). If you confirm the latest commit fixes the issue I'll release 0.10.3 with this fix immediately.

pstolarz, 2022-03-08:
> @homonto lepiej kanal na gitter.im zalozyc
Dobry pomysł. Można tam pogadać nieklawiaturowo też?

pstolarz, 2022-03-08:
Założyłem grupę OneWireNg na tym gitterze. Jakby co to zapraszam. Później dodam jeszcze info o tej grupie na README. Dzięki za pomysł.

uzi18, 2022-03-09:
dodaj takie cos: 
[image]

pstolarz, 2022-03-09:
@homonto @uzi18 I've just released a new 0.10.3 version of the lib with bit-banging timings set to strict by default. It will take some time to sync the lib with the lib-repos (PlatformIO, Arduino). I'd be grateful if you could check the new version on your ESP32 setups with WiFi communications and let me know about its stability now. Thanks for the catch!

homonto, 2022-03-09:
you mean: shut down all my RPi? ;-)
I will check and will let you know