pstolarz/OneWireNg #39: __has_include(<new>) not working on ESP8266 Platformio builds using old framework

thorrak, 2022-03-20:
Hey there,

When I attempt to compile the [DallasTemperature.ino ](https://github.com/thorrak/OneWireNg/blob/master/examples/DallasTemperature/DallasTemperature.ino) example for an ESP8266 using `platform = espressif8266@2.6.3` (the latest version of the 2.x ESP8266 platform available on Platformio), I get the following error:

```
In file included from .pio/libdeps/esp12e/OneWireNg/src/OneWireNg.h:19:0,
                 from .pio/libdeps/esp12e/OneWireNg/src/OneWireNg_BitBang.h:16,
                 from .pio/libdeps/esp12e/OneWireNg/src/OneWireNg_BitBang.cpp:15:
.pio/libdeps/esp12e/OneWireNg/src/platform/Platform_New.h: In function 'void* operator new(size_t, void*)':
.pio/libdeps/esp12e/OneWireNg/src/platform/Platform_New.h:52:14: error: redefinition of 'void* operator new(size_t, void*)'
 inline void *operator new(size_t sz, void *ptr) NOEXCEPT {
              ^
In file included from /Users/jbeeler/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_construct.h:59:0,
                 from /Users/jbeeler/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_tempbuf.h:60,
                 from /Users/jbeeler/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:62,
                 from /Users/jbeeler/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/Arduino.h:238,
                 from .pio/libdeps/esp12e/OneWireNg/src/platform/Platform_Delay.h:17,
                 from .pio/libdeps/esp12e/OneWireNg/src/OneWireNg_BitBang.cpp:13:
/Users/jbeeler/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/new:109:14: error: 'void* operator new(std::size_t, void*)' previously defined here
 inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
              ^
```

Digging into it, it appears that the `__has_include` [check in Platform_New.h](https://github.com/pstolarz/OneWireNg/blob/master/src/platform/Platform_New.h#L30-L33
) is failing. 

Best that I can tell `__has_include` is #define'd as `0` in this version of the xtensa toolchain, meaning that simply testing if it is defined is not proof-positive that it actually available. 

I've produced a MCVE here: https://github.com/thorrak/onewireng_platform_new_issue -- `[env:esp8266_2.6.3]` is broken, while `[env:esp8266_current]` works.

thorrak, 2022-03-20:
To help explain why I need this to work on the v2.x platform line, the v3.x platform line has a bug in the WiFi binaries that causes a race condition with my router when attempting to connect to WiFi, making it so that the closer the device is to the router, the less likely it is to actually connect. Very strange (and especially problematic given that the framework the ESP8266 Arduino codebase is based upon is no longer being supported by Espressif, and therefore the bug is unlikely to get fixed)

thorrak, 2022-03-20:
I just submitted #40 which provides a workaround to this issue. Not ideal, but it is in line with the existing method to force the alternate implementation.

closed 2022-03-20 by pstolarz

thorrak, 2022-03-20:
I can confirm that solves it. Thank you!

pstolarz, 2022-03-20:
Thanks for the catch, the <new> header is really nightmare with enormous world of various toolchains.
I decided to use some more elegant method than providing artificial param as proposed by you.

How do you like of the lib on ESP8266? The platform is loosing its popularity in the advent of ESP32.

PR #41 (2022-03-20): Platform_New.h header fix
Fixes #39