pstolarz/OneWireNg #44: Issues with DS18B20

indymx, 2022-04-07:
I'm having an issue where after a random amount of time the DS18B20 will stop sending valid data. I don't have any good debug showing what data is being sent or if any when the issue starts, but I do have my script go into a failure loop when it see's a 0 for temp on the sensor. 

It usually requires a complete power cycle of the ESP32 to get it to read again. Is there a way to force the sensor to reset ?

pstolarz, 2022-04-07:
* Does this occur for OneWireNg only or for OneWire too?
* What version of the lib do you use?
* What type of libs do you use? OneWireNg only or, for example, OneWireNg + OneWire bridge + [Arduino-Temperature-Control-Library](https://github.com/milesburton/Arduino-Temperature-Control-Library) lib? If the 2nd case - do you observe the same behavior for pure OneWireNg lib with native [DallasTemperature](https://github.com/pstolarz/OneWireNg/blob/master/examples/arduino/DallasTemperature/DallasTemperature.ino) example?
* Does after reaching the failed state a device is not able to work correctly until its reset? maybe some Chinese counterfeits?

pstolarz, 2022-04-07:
I need to know exact version. There is a problem with 0.10.2 on ESP32. Use 0.10.3 or 0.11.0 (preferred). Check with the DallasTemperature example if the problem occurs.

indymx, 2022-04-07:
Just checked, I'm showing 0.10.3 in the library.properties file. 

I'll update to 0.11 tonight when I get home.

pstolarz, 2022-04-07:
I run the DallasTemperature example on my ESP32-S2 setup for about 1 hour and didn't get any single error for temp. reading.

indymx, 2022-04-07:
This time it ran for about 7 hours before it stopped working.

indymx, 2022-04-07:
It seems that authentic Dallas / Maxim Integrated sensors are impossible to get right now.

pstolarz, 2022-04-07:
You may consider push-pull circuit controlled by dedicated GPIO which will reset the DS18B20 (shorts its VDD pin to GND) in case of such deadlock.

https://open4tech.com/open-drain-output-vs-push-pull-output/

pstolarz, 2022-04-07:
https://github.com/cpetrich/counterfeit_DS18B20

indymx, 2022-04-07:
I will have to do some research on the push/pull. I read that link, I understand the concept, but need to see exactly how to implement that.. 

Second link is awesome. Thanks for sharing. That will definitely come in handy.

indymx, 2022-04-08:
Well, it appears that I am dealing with Family C clones. 28-FF-64-1E-08-75-19-06: Family C (Clone).

Since it's near impossible to get genuine sensors right now, do you have any suggestions beyond the push/pull cirucuit?

pstolarz, 2022-04-08:
The circuit is very simple and consists of 2 MOSFETS (P and N)

## Work mode
[image]

GPIO is set to low (GND), therefe Q1 is conducting Q2 is not. DS18B20's VDD pin is provided with voltage.

## Reset mode
[image]

GPIO is set to High (Vcc), therefe Q1 is not conducting and VDD is shorted to GND via Q2. DS18B20 is not powered in this case (reset state).

The question is if it's really a problem with DS18B20 and it's reset will solve your issue.

pstolarz, 2022-04-08:
If you would like to experiment with this circuit I recommend this simulator:

https://www.falstad.com/circuit

pstolarz, 2022-04-08:
At first try you may to use GPIO (in the output mode) connected directly do DS18B20's VDD pin to power it. In working mode GPIO is high, for reset low. Maybe it will be sufficient for your case and you dont need to bother with push-pull circuit.

indymx, 2022-04-08:
I'm not finding anything in the datasheets that explicitly says whether or not it's open-drain.

I'm using an Adafruit ESP32S2 TFT. The ESP32-S2 datasheet has this: (but that's the only reference I can find)

[image]

pstolarz, 2022-04-08:
Connect DS18B20 as follows:
[image]

* During initialization phase `GPIO_VDD` is set high (output, non open-drain mode), `GPIO_DQ` is managed by the lib:
```
pinMode(GPIO_VDD, OUTPUT);
digitalWrite(GPIO_VDD, 1);
new (&_ow) OneWireNg_CurrentPlatform(GPIO_DQ, false);
```

* For reset `GPIO_VDD` and `GPIO_DQ` are shorted to GND for specific amount of time (1 sec here):
```
auto& ow = (OneWireNg_CurrentPlatform&)_ow;

digitalWrite(GPIO_VDD, 0);
ow.setBus(0);
delay(1000);
```
NOTE: `OneWireNg_BitBang::setBus()` method is protected so you need to move this method from protected to public in `OneWireNg_BitBang.h` file

* Back to life after reset. `GPIO_VDD` to high. `GPIO_DQ` set to high impedance (as input):
```
auto& ow = (OneWireNg_CurrentPlatform&)_ow;

ow.setBus(1);
digitalWrite(GPIO_VDD, 1);
```

closed 2022-04-10 by pstolarz