pstolarz/OneWireNg #45: Arranging ID's or setting ID's

vangalvin, 2022-04-13:
A heads up that OneWireNG works perfectly on the UM Feather S2.

For my project I have a pretty long run of DS18B20's The test run is 35 long but the final version that will be deployed is going to be about 100 sensors long. What is the easiest way to assign an ID to each DS18B20 so I get sensor 1 at the top and 100 at the end and dont have to worry about arranging them when im soldering them inline?

pstolarz, 2022-04-14:
I don't know such method. There are sensors like DS1825, which support A0-A3 pins uniquely identifying sensor locations on a bus but the number is constrained to 16.

closed 2022-04-15 by pstolarz

uzi18, 2022-04-15:
@pstolarz  Maybe it could be implemented as new optional feature - IDs cacheing during bus search? Plus event for new device detected?

pstolarz, 2022-04-15:
@uzi18 So, what's the difference in storing such table in user's code as C++ std::map for example? On the other hand, it's difficult to foresee all possible demands, a lib user may have in such cases and fulfill them on the lib level. For me this is strictly user code functionality. [KISS](https://en.wikipedia.org/wiki/KISS_principle) in other words.

vangalvin, 2022-04-15:
> @uzi18 For me this is strictly user code functionality.

I think you are right except I may do this on thye users end. Data from the sensors is being packed then sent via Swarm Satelite to the research station who can unpack it. when they import it it will sort the sensor readings in to the correct order. This will save a bit of power as the unit is battery/solar and its dark down there 6 months of the year. 
The faster i can get the readings the quicker i can put the unit back to sleep.

pstolarz, 2022-04-15:
Even worse, they could be sent to Antipodes, where everything go upside down and lots of energy is consumed on moving forward... If you know what I mean 😜

vangalvin, 2022-04-15:
Hahaha :) IDK, Mt Erebus is enough of a challange :)