pstolarz/OneWireNg #49: Question: Read a specific sensor

DerDoubleD, 2022-06-06:
Is it possible to read only one specific sensor?
Example:
I know that the sensor for BadRoom has the ID 28:FF:2C:A4:B3:16:3:CA and for LivingRoom 28:FF:D5:98:B3:16:3:C5.
Is it possible to do something like 
BadRoom = getTemp(28:FF:2C:A4:B3:16:3:CA)
LivingRoom = getTemp(28:FF:D5:98:B3:16:3:C5)

Thanks!

pstolarz, 2022-06-06:
Yes it's possible. `DSTherm::readScratchpad()` accepts sensor id at its argument. See [DallasTemperature.ino](https://github.com/pstolarz/OneWireNg/blob/master/examples/arduino/DallasTemperature/DallasTemperature.ino) example for details.

DerDoubleD, 2022-06-06:
Thanks for the Answers.
Sorry i can not find an example with this and i am not good enough to find it out by my self. 

DSTherm::readScratchpad(28:FF:2C:A4:B3:16:3:CA)
Is this right?

Thanks!

pstolarz, 2022-06-06:
```c
#include "OneWireNg_CurrentPlatform.h"
#include "drivers/DSTherm.h"
#include "utils/Placeholder.h"

#define OW_PIN          13

static OneWireNg *ow;

void setup() {
    Serial.begin(115200);
    ow = new OneWireNg_CurrentPlatform(OW_PIN, false);
}

void loop() {
    OneWireNg::Id id = {0x28, 0xFF, 0x2C, 0xA4, 0xB3, 0x16, 0x03, 0xCA};

    DSTherm drv{*ow};
    Placeholder<DSTherm::Scratchpad> _scrpd;

    drv.convertTemp(id, DSTherm::SCAN_BUS);

    if (drv.readScratchpad(id, &_scrpd) == OneWireNg::EC_SUCCESS)
    {
        float temp = ((DSTherm::Scratchpad&)_scrpd).getTemp() / 1000.0;

        Serial.print(temp);
        Serial.print(" C");
        Serial.println();
    }
    delay(1000);
}
```