pstolarz/OneWireNg #57: Arduino IDE

willishf, 2023-01-13:
Trying to test OneWire on raspberry pico W in Arduino IDE 2.0.3. Appears to be a problem others are having when downloading code the IDE hangs with an uploading message that doesn't go away. https://forum.arduino.cc/t/often-stuck-on-uploading-ide-2-0-3/1063796/4

For OneWireNg Arduino testing on Pico W what version of Arduino IDE are you using?

Larger goal is to add OneWire support in Firmata to abstract out the need to write code on the pico W device in the field and push it up to server code. The low cost wireless of Pico W opens up some interesting and scaleable remote sensor projects. Want to avoid pushing out new code every time a new sensor gets added.

closed 2023-01-13 by pstolarz

willishf, 2023-01-13:
> IDE version is irrelevant here. What is relevant is the Arduino core used
> for the build. I'm always using the latest stable core.
> In case of OneWire problems, please report them to the OneWire repository.
> I'm not responsible the issues there.

Testing OneWireNG code on pico W get the same uploading error that others
are reporting with a range of test code on arduino ide 2.0.3 . Wanted to
see if your testing/development on Arduino 1.8 IDE. Trying to avoid having
multiple problems when trying to get OneWire code wiring on pico W.

pstolarz, 2023-01-13:
Have you looked this post:
https://arduino.stackexchange.com/questions/91685/ds18b20-temperature-sensor-crashes-mbedos-on-arduino-nano-rp2040

pstolarz, 2023-01-13:
FYI: I've just compiled and uploaded OneWIreNg's DallasTemperature example using IDE 2.0.3 on my setup (RP2040 Pico board). Core used: earlephilhower/arduino-pico. Works fine.

willishf, 2023-01-13:
Thanks for the reference on uninstalling onewire library. Will try that.