pstolarz/OneWireNg #59: [Question] Can this library be used to implement half-duplex communication between 2 controllers?

Roming22, 2023-10-09:
I have 2 boards, connected with 3 wires (power, ground, data). I'm trying to pass messages back and forth.

Would this library allow me to do that? If so, how would I configure the controller that acts as the "OneWire device"?

pstolarz, 2023-10-09:
Basically not. 1-wire is a proprietary Dallas protocol for use with their devices. It works in master-slave mode and is not intended as a general purpose protocol to transmit generic data.

Some time ago I had similar problem while working with ESP-01 board (ESP8266 MCU). It has limited number of GPIOs  therefore I needed to connect it to an external AVR MCU to extend its functionality. I've invented proprietary (only dedicated for my application), CRC-protected, single-data-wire protocol working in half-duplex mode with ESP as master (Tx) and AVR as slave (Rx) to control AC relays via dedicated AVR's GPIOs.

pstolarz, 2023-10-09:
This lib handles only 1-wire master part. To handle device part you will need to use slave library. I'm aware about [OneWireHub](https://github.com/orgua/OneWireHub) which emulates large number of 1-wire slaves, but I don't have experience with it.

closed 2023-10-09 by pstolarz