pstolarz/OneWireNg #63: Library cannot work with Arduino Opta

MarlinNano, 2024-05-06:
I tried connecting a DS18B20 sensor to an Arduino Opta using this library. The problem is that the OneWire protocol is half duplex, while the Arduino Opta only has digital **inputs**, not outputs (the only outputs it has are the four relays). 
[Arduino Opta datasheet](https://docs.arduino.cc/resources/datasheets/AFX00001-AFX00002-AFX00003-datasheet.pdf)

pstolarz, 2024-05-07:
Yes, it looks like you are right. The device doesn't offer general purpose GPIOs.

closed 2024-05-07 by pstolarz

uzi18, 2024-05-08:
What about aux connector?

MarlinNano, 2024-08-18:
> What about aux connector?

I think it will be possible using the AUX connector, however until now I don't think Arduino has released a 1-Wire expansion board for the OPTA. I found this cool little breakout board someone made that enabled serial and I2C for the OPTA, here is the [link](https://www.hackster.io/news/arduino-opta-serial-and-i2c-breakout-a74dfb9f414e).

pstolarz, 2024-08-19:
Looking at the Opta [schematics](https://docs.arduino.cc/hardware/opta) I see some IN_x pins are connected directly to STM32H PFx pins which serve as generic I/O. So what's the problem to use them for 1 write bit banging?