pstolarz/OneWireNg #67: Unknown address of the DS18B20 temp sensor

upeer, 2024-09-12:
Hello, I am using TTGO Lora 32 Oled with a few DS18B20 temperature sensors. My code uses DallasTemperature 3.9.0 library na OneWireNG 0.13.3 (I was not able to compile the code when other OneWire library used).
I have 60 pieces of DS18B20 sensors and 6 of them return UNKNOWN address, and about 10 of them return incomplete address - for example UNKNOWNf8 etc. What I am doing wrong or is there some error in the library?
Thank you!

pstolarz, 2024-09-12:
Try to run DallasTemperature example from this lib and observe results on your setup.

pstolarz, 2024-09-12:
Additionally you may try this [app](https://github.com/cpetrich/counterfeit_DS18B20) to check if your sensors are not just cheap Chinese counterfeits.

upeer, 2024-09-13:
Thank you for the quick response!

I tried the example and it reads the address from all the sensors.
It looks that the problem is, when somewhere in the address is just one character.
For example when the address is 28:61:64:34:E8:31:E5:B (just the B at the end), then my code returns UNKNOWN.
In other case when the single character is in the middle of the address, my code returns just a part of address. For example 28:61:64:34:89:47:1:E5 returns just E5 (stops on 1)

Problem is probably in my code:

String getAddress(DeviceAddress devAdr, uint8_t idx) {
  String adr = "";
 
  for (uint8_t i = ADR_START; i < ADR_END; i++) {
    if (devAdr[i] < 16) {
      adr = "UNKNOWN";
    } else {
      adr = adr + String(devAdr[i], HEX);
    }
  }

  return adr;
}

uzi18, 2024-09-13:
Show more code related to addresses, declaratiom, getting, printing etc.

uzi18, 2024-09-13:
@upeer  Try this:
```
String getAddress(DeviceAddress devAdr, uint8_t idx) {
String adr = "";

for (uint8_t i = ADR_START; i < ADR_END; i++) {
  if (devAdr[i] < 16) {
    adr += "0";
  } else {
    adr += String(devAdr[i], HEX);
  }
}

return adr;
}
```

pstolarz, 2024-09-13:
@upeer you may look at `printId()` from `DallasTemperature.ino` how to print an 1-wire address.
I'm closing the issue as unrelated to the lib.

closed 2024-09-13 by pstolarz

uzi18, 2024-09-13:
https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/oneWireSearch/oneWireSearch.ino
Example code

upeer, 2024-09-16:
> @upeer Try this:
> 
> ```
> String getAddress(DeviceAddress devAdr, uint8_t idx) {
> String adr = "";
> 
> for (uint8_t i = ADR_START; i < ADR_END; i++) {
>   if (devAdr[i] < 16) {
>     adr += "0";
>   } else {
>     adr += String(devAdr[i], HEX);
>   }
> }
> 
> return adr;
> }
> ```

> @upeer Try this:
> 
> ```
> String getAddress(DeviceAddress devAdr, uint8_t idx) {
> String adr = "";
> 
> for (uint8_t i = ADR_START; i < ADR_END; i++) {
>   if (devAdr[i] < 16) {
>     adr += "0";
>   } else {
>     adr += String(devAdr[i], HEX);
>   }
> }
> 
> return adr;
> }
> ```

Thank you for your tips. I only changed the code to next one and now it works...
String getAddress(DeviceAddress devAdr, uint8_t idx) {
  String adr = "";
 
  for (uint8_t i = ADR_START; i < ADR_END; i++) {
    String adPart = String(devAdr[i], HEX);
    if (2 > adPart.length()) {
      adr = adr + "0" + adPart;
    } else {
      adr = adr + adPart;
    }
  }

  return adr;
}