pstolarz/OneWireNg #68: Fails to build on Raspberry Pi Pico2W

tyeth, 2024-11-27:
Hey, there appears to be an issue building for RP2350 based Pico2W.
This section doesn't seem correct, it prioritises the ARDUINO ifdef over the PICO_BUILD, which results in the error:
```
.pio\libdeps\pimoroni_pico_plus_2w\OneWireNg\src\platform\Platform_TimeCritical.cpp:19:1: error: 'tc_t' does not name a type
   19 | tc_t _tc[portNUM_PROCESSORS];
      | ^~~~
*** [.pio\build\pimoroni_pico_plus_2w\lib610\OneWireNg\platform\Platform_TimeCritical.cpp.o] Error 1
```
https://github.com/pstolarz/OneWireNg/blob/master/src/platform/Platform_TimeCritical.h#L104-L110

If I move the arduino/mbed ifdef checks after pico_build then the compilation completes as expected.

(I'm building against the pimoroni version of the rp2350 + RP2 wifi module as the [pico2w hasn't made it into the board definitions yet for MaxGerhardt's platformio repo](https://github.com/maxgerhardt/platform-raspberrypi/issues/86).)

closed 2024-11-27 by pstolarz

pstolarz, 2024-11-27:
Beside this does the lib work for the platform in question?

tyeth, 2024-11-27:
It appears to, I've admittedly not got to testing properly yet, just got the board to connect to adafruit IO (used for the wippersnapper firmware). Tomorrow I'll finish the board definition so the pins align, and then test the ds18b20 code which relies on this library.

Thanks for the great work!

tyeth, 2024-11-27:
The power of open-source 😁 
We found various issues with the main onewire along with not having releases, I'd been playing with this one occasionally instead, and I think one esp-idf change made us take the plunge and swap over for the Wippersnapper project.

PR #69 (2024-11-27): Pico SDK build fix
Fixes #68