pstolarz/OneWireNg #71: Maximum number of OneWireNg_CurrentPlatform objects?

Ing-Dom, 2025-01-25:
Hi!

Thanks for that great lib.
I recently ran into a problem.. I think when I create more then 2 OneWireNg_CurrentPlatform objects
`new (&m_ow) OneWireNg_CurrentPlatform(pin0, false);`
my programm chrashes. 2 are fine.

Is this a limitation?

due to the wiring of my project (star topology) I would ne up to 16..

Running on RP2040 with arduino-pico core (within a PlattformIO project..)
for reference
https://github.com/OpenKNX/OFM-THPSensorModule/blob/v1/src/HWSensorchannel_DS18B20.cpp

pstolarz, 2025-01-26:
Hi @Ing-Dom, yes for RP2040 there is a limit for number of 1-wire masters with PIO driver (the default driver for the platform). Please take a look here for datails: https://github.com/arduino/ArduinoCore-mbed/issues/316#issuecomment-1455046299

Ing-Dom, 2025-01-26:
ok, thank you for your reply.
Im now a little bit puzzled, because I thought - based on the documentation which I also understodd wrong - that also on RP2040, Bitbang is the default.
But - shame on me - I didn't do check the code..

https://github.com/pstolarz/OneWireNg/blob/c4d21d0191567ae0aaddb22931480f71ac87fdef/src/OneWireNg_Config.h#L332

you stated

> Using OneWireNg's OneWireNg_PicoRP2040PIO (1-wire PIO driver), it's possible to create up to 4 different 1-wire drivers working on 4 different GPIO pins - 2 of them on PIO-0 and 2 on PIO-1 (provided no other PIO program is already uploaded). PIO number for use by the driver is passed by 3rd argument of OneWireNg_PicoRP2040PIO constructor (by default PIO-0 is used).

Do I understand correctly:
When I need more than 2 drivers, I need to create the OneWireNg_PicoRP2040PIO myself, because I need to pass PIO-1 to number 3+4, right?
Is that the reason it crashes?

Anyway, I need to poll 16 DS18B20, connected to 16 different GPIO.
But I don't need to do this in parallel - so I could re-use one or 2 drivers to do that - but on a different pin.
right? Any advice how to do that?

pstolarz, 2025-01-26:
Yes, you need to create 4 OneWireNg_PicoRP2040PIO objects 2 of them with 0 as 3rd arg for the class constructor and 2 with 1. See  https://github.com/pstolarz/OneWireNg/blob/57d47c175f46de5f0805ea5c5854084b6410380f/src/platform/OneWireNg_PicoRP2040PIO.h#L55

What I don't understand it's why you need 16 masters to handle 16 sensors, instead having them all connected to single wire controlled by a single master.

Ing-Dom, 2025-01-26:
because my hardware main purpose was to be able to connect up to 8 SHT3x (or other I2C Sensors) which only can be used once on one Bus.
Later there was the idea to also allow DS18B20  - and having only one sensor per GPIO make you free from assigning Sensor IDs..
thats the hardware:
https://github.com/OpenKNX/OpenKNX/wiki/SEN-UP1-8xTH

Is it possible to re-assign a different pin to a specific driver in programm flow?

pstolarz, 2025-01-26:
Yes you may destroy the driver na calling it's destructor and recreating it by calling new in-place constructor with different GPIO.

Ing-Dom, 2025-01-26:
I just wanted to answer you, that that would have been my first guess also, but there is no destructor :)
But you discovered yourself in the meantime.

I'll check the bitbang drivers in the meantime and thanks for your help ! Really appreciate it.

pstolarz, 2025-01-26:
I don't recommend the BB driver for RP2040, the PIO one is much better. I'm gonna do some fixes for the PIO driver soon and give you for tests.

Ing-Dom, 2025-01-26:
ohhhh :) that was fast, thanks, Ill give it a try immidiatelly

Ing-Dom, 2025-01-26:
```
        Placeholder<OneWireNg_PicoRP2040PIO> m_ow;
        Placeholder<OneWireNg_PicoRP2040PIO> m_ow2;
  
 new (&m_ow) OneWireNg_PicoRP2040PIO(pin0, false);
    new (&m_ow2) OneWireNg_PicoRP2040PIO(pin1, false);

    DSTherm drv(m_ow);
    drv.writeScratchpadAll(0, 0, DSTherm::RES_12_BIT);

    DSTherm drv2(m_ow2);
    drv2.writeScratchpadAll(0, 0, DSTherm::RES_12_BIT);

    delete (&m_ow); // <- that crashes. Is that correct? Im not so familiar what you are doing here with the template class..
```

pstolarz, 2025-01-26:
No exactly, if you are using in-place driver creation:

```cpp
Placeholder<OneWireNg_PicoRP2040PIO> ow;

new (&ow) OneWireNg_PicoRP2040PIO(pin0, false);

DSTherm drv(ow);
// ...

// calls destructor in-place w/o actual memory deallocation
(*ow).OneWireNg::~OneWireNg();
```

for classing allocation/deallocation case:

```cpp
OneWireNg *ow = new OneWireNg_PicoRP2040PIO(pin0, false);

DSTherm drv(*ow);
// ...

// delete the object
delete ow;
```

Ing-Dom, 2025-01-26:
hmm there seems to be some problem.
I poll one sensor after the other - this way:

```
{
m_ow0 = new OneWireNg_PicoRP2040PIO(m_pin0, false);
DSTherm drv0(*m_ow0);
drv0.convertTempAll(0, false);
}

delay..

{
DSTherm drv0(*m_ow0);

static Placeholder<DSTherm::Scratchpad> scrpd;
OneWireNg::ErrorCode ec = drv0.readScratchpadSingle(scrpd);
if (ec == OneWireNg::EC_SUCCESS)
{
	long temp = scrpd->getTemp2();
	SetTemperature((float)temp / 16);
}
else if (ec == OneWireNg::EC_CRC_ERROR)
{
	logDebugP("CRC-Error 0");
}
else
{
	logDebugP("Onewire Error 0x%x",ec);
}

drv0.~DSTherm();
m_ow0->~OneWireNg_PicoRP2040PIO();
}
```

the first two sensor - regardless which pin - work.
all other sensors / pins return a crc error.
when i reach the first two again, they work, all others don't.

pstolarz, 2025-01-27:
Thanks, I will look at this later today...

pstolarz, 2025-01-27:
I've just tested the fix on my setup (RP2040, PicoSDK lib) and works as expected. Code as follows:

```cpp
static void printScratchpad(const DSTherm::Scratchpad& scrpd)
{
    const uint8_t *scrpd_raw = scrpd.getRaw();

    printf("  Scratchpad:");
    for (size_t i = 0; i < DSTherm::Scratchpad::LENGTH; i++)
        printf("%c%02x", (!i ? ' ' : ':'), scrpd_raw[i]);

    printf("; Th:%d; Tl:%d; Resolution:%d",
        scrpd.getTh(), scrpd.getTl(),
        9 + (int)(scrpd.getResolution() - DSTherm::RES_9_BIT));

    long temp = scrpd.getTemp2();
    printf("; Temp:");
    if (temp < 0) {
        temp = -temp;
        printf("-");
    }
    printf("%d.%04d C\n", (int)temp / 16, (10000 * ((int)temp % 16)) / 16);
}

int main()
{
    stdio_init_all();

    Placeholder<DSTherm::Scratchpad> scrpd;

    for(int i = 0;; i++) {
        int pin = (i % 16);
        printf("[PIN %d]\n", pin);

        auto ow = OneWireNg_CurrentPlatform(pin, false);
        DSTherm drv(ow);

        drv.convertTempAll(DSTherm::MAX_CONV_TIME, false);

        for (const auto& id: ow) {
            OneWireNg::ErrorCode ec = drv.readScratchpad(id, scrpd);
            if (ec == OneWireNg::EC_SUCCESS) {
                printScratchpad(scrpd);
            } else {
                printf(" ERROR: %d\n", ec);
            }
        }
    }

    return 0;
}  
```

so, I'm probing GPIOs 0-15 in the main loop against connected sensors printing readouts on the console. Try this code on your setup.

Ing-Dom, 2025-01-27:
I had to modify you code a little bit to run within my arduino-pico environment.

```
#include <stdio.h>
#include "pico/stdio.h"

#include "OneWireNg_CurrentPlatform.h"
#include "drivers/DSTherm.h"
#include "utils/Placeholder.h"
#include "platform/Platform_Delay.h"
#include "DallasTemperature.h"

static void printScratchpad(const DSTherm::Scratchpad& scrpd)
{
    const uint8_t *scrpd_raw = scrpd.getRaw();

    Serial.printf("  Scratchpad:");
    for (size_t i = 0; i < DSTherm::Scratchpad::LENGTH; i++)
        Serial.printf("%c%02x", (!i ? ' ' : ':'), scrpd_raw[i]);

    Serial.printf("; Th:%d; Tl:%d; Resolution:%d",
        scrpd.getTh(), scrpd.getTl(),
        9 + (int)(scrpd.getResolution() - DSTherm::RES_9_BIT));

    long temp = scrpd.getTemp2();
    Serial.printf("; Temp:");
    if (temp < 0) {
        temp = -temp;
        Serial.printf("-");
    }
    Serial.printf("%d.%04d C\n", (int)temp / 16, (10000 * ((int)temp % 16)) / 16);
}

void setup()
{

    Placeholder<DSTherm::Scratchpad> scrpd;
    uint8_t mypins[4] = {22,23,24,25};

    Serial.begin();
    while(!Serial)
        delay(1);
    Serial.println("Start reading DS18B20..");
    delay(1000);
    Serial.println("Start reading DS18B20..");
    delay(1000);
    Serial.println("Start reading DS18B20..");
    delay(1000);

    for(int i = 0;; i++) {
        int pin = mypins[i % 4];
        Serial.printf("[PIN %d]\n", pin);

        auto ow = OneWireNg_CurrentPlatform(pin, false);
        DSTherm drv(ow);

        drv.convertTempAll(DSTherm::MAX_CONV_TIME, false);

        for (const auto& id: ow) {
            OneWireNg::ErrorCode ec = drv.readScratchpad(id, scrpd);
            if (ec == OneWireNg::EC_SUCCESS) {
                printScratchpad(scrpd);
            } else {
                Serial.printf(" ERROR: %d\n", ec);
            }
        }
    }
}

void loop()
{}
```

it prints;

> [PIN 22]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 23]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 24]
>   Scratchpad: 74:01:00:00:7f:ff:7f:10:4e; Th:0; Tl:0; Resolution:12; Temp:23.2500 C
> [PIN 25]
>   Scratchpad: 70:01:00:00:7f:ff:7f:10:5b; Th:0; Tl:0; Resolution:12; Temp:23.0000 C
> [PIN 22]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 23]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 24]
>   Scratchpad: 74:01:00:00:7f:ff:7f:10:4e; Th:0; Tl:0; Resolution:12; Temp:23.2500 C
> [PIN 25]
>   Scratchpad: 70:01:00:00:7f:ff:7f:10:5b; Th:0; Tl:0; Resolution:12; Temp:23.0000 C
> [PIN 22]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 23]
>   Scratchpad: 77:01:00:00:7f:ff:7f:10:8b; Th:0; Tl:0; Resolution:12; Temp:23.4375 C
> [PIN 24]
>   Scratchpad: 74:01:00:00:7f:ff:7f:10:4e; Th:0; Tl:0; Resolution:12; Temp:23.2500 C
> [PIN 25]
>   Scratchpad: 70:01:00:00:7f:ff:7f:10:5b; Th:0; Tl:0; Resolution:12; Temp:23.0000 C
> 

so, it seems to work if you do it the right way. I'll try again to integrate into my setup..

tyeth, 2025-01-27:
Wow, what timing, we have a bug related to destroying and reinitialising the pio driver (user adds DS18b20 device and we initialise the bus - works 1st and 2nd time and crashes 3rd), I was going to suggest implementing the destructor, but you did yesterday!

Looks like I missed turning off the outputs (instead just unclaiming them). I'll retest with your PR.

pstolarz, 2025-01-28:
@tyeth Let me know about results. I'm going to merge #72 later today, do you need a new OneWireNg release with the change?

PR #72 (2025-01-28): OneWireNg_PicoRP2040PIO: destructor added
Fixes #71
closed 2025-01-28 by pstolarz