pstolarz/OneWireNg #73: Can not use setThl to write the alam registers

lit1s, 2025-02-14:
Dear Sir. I've managed to work with you library oneWireNg, with an RP2040, earlephilhower core, on vscodium. I'm not very used to C++, but I've managed to make work almost all:

I can read temperature, and can read alarm registers, with this snips, as in your examples: (tl register, with a similar function get Tl1

int32_t getTemp1 (const DSTherm::Scratchpad& scrpd1)
{
    int32_t temp = scrpd1.getTemp ();
    return temp;
}

int8_t getTh1  (const DSTherm::Scratchpad& scrpd1)
{
    int8_t data = scrpd1.getTh ();
    return data;
}

But I cannot use the writeScratchpad () function, can not manage to compile! Can you give us an example?

pstolarz, 2025-02-14:
You may find `writeScratchap()` use in

closed 2025-02-14 by pstolarz

lit1s, 2025-02-15:
Only this two lines:

drv1.writeScratchpadAll (th, tl, COMMON_RES);
drv1.copyScratchpadAll ();

And finally worked, thanks a lot! :-)

uzi18, 2025-02-15:
You can use resolution from scratchpad